modelica-3rdparty/PowerSystems #14: Example simulation error

hjw19, 2016-02-17:
Error occurs when i simulate 'PowerSystems/Examples/GenerationAC3ph/Turbin generator' example file.

Error message is as follows:
======================================================
division by zero at time 0, (a=45160) / (b=0), where divisor b expression is: generator.c.omega_nom

Debug more

simulation terminated by an assertion at initialization

Process crashed

Simulation process failed. Exited with code -1.
=====================================================
Version of modelica is 'OpenModelica-v1.9.4-dev.beta1-85'
What's the problem?

tbeu, 2016-03-23:
I guess this issue rather should go to the OMC developers, e.g. @rfranke @sjoelund

sjoelund, 2016-03-24:
For OMC issues go to trac.openmodelica.org/OpenModelica

rfranke, 2016-03-26:
PowerSystems, like many other Modelica libraries, uses Dymola as the reference tool.

OpenModelica has nightly tests for the PowerSystems library. You see that so far only around 77% of the models compile and 69% simulate:
   https://test.openmodelica.org/libraries/PowerSystems/BuildModelRecursive.html

There are also some active support tickets. Search for PowerSystems under
   https://trac.openmodelica.org/OpenModelica/report

closed 2016-03-26 by rfranke