modelica-3rdparty/PowerSystems #18: Inertial frame not working on generic components?

ceraolo, 2016-10-27:
I've seen that switching refType from inertial to synchronous changes signals (e.g. voltages) from oscillating to constant (when systems are in steady state), that is totally satisfactory: it is like switching from Clarke to Park transform.

I tried some PowerFlow.GenericComponentsTest models, also. In these cases, instead the quantities remain constant (i.e. do not oscillate) in inertial mode, which seems to be wrong.

Take for instance FixedLoadTest: switching from Synchron to Inertial does not change anything in the shape of signals.
Is this an issue or there is something I'm disregarding?
I tried this on the 2016-06-11 version of the library.

rfranke, 2016-10-28:
You are right. The generic components always use synchronous mode. This is not nice if the system model enables to select inertial mode and nothing happens. It might be possible to enhance the generic components, in particular since PhaseSystem_dq got a second reference angle like PhaseSystem_dq0: https://github.com/modelica/PowerSystems/commit/938b3109e41bcd6414e438c35e7626a495da63a8#diff-9445c81db99bb66f80db29e46fa96c4b

ceraolo, 2016-10-28:
I tried to dig on the PhaseSystem structure and got some insight on how it works. Unfortunately not everything is sufficiently clear to me to attempt modifications.
Maybe, in case  I understand things better, I could try to do this enhancement on generic components, that would render the whole library more consistent, in the future.
Unless some other person, more expert than me on the inner structure of Power Systems Library, does this earlier...

ceraolo, 2016-10-29:
To give my contribution I need some very basic info.
The Park transform that I use makes usage of a single theta angle:
[image]
This is able to deal with generic three-phase quantities: they don't need to be a balanced system, they don't even need to be sinusoidal.
Power Systems Library must use a different theory, since it requires two angles. I could do some reverse engineering of the library, but is a too heavy task for me (and the available time I have).
Can you, please, address me to some reference where I can find the underlining theory of this two angles transformation?
Explanation of how Park transform is used in PowerSystems library, and definition of P and Q, and of the several "j" operators, would be very much welcome as well.

I've read the paper of yours 
  Flexible modeling of electrical power systems - the Modelica PowerSystems library 
that is very helpful, but stil not sufficient to my purposes.
Thanks in advance.

rfranke, 2016-11-01:
That's exactly the beauty of Park's transformation: two constant quantities are sufficient for the balanced stationary case while unbalanced systems and transients are still covered. 

The two angles are just to distinguish between Synchonous and Inertial mode. See
a8037d65faa7ebc84d81a9e4fc62d78ab0678303 that provides the inertial frame for Generic components.

ceraolo, 2016-11-02:
Thanks a lot for you explanation and  you work on the library.
I will stay tuned!

closed 2016-11-02 by rfranke

ceraolo, 2016-12-02:
Ehm,
I suppose that inertial frames, when available should work.
I've found that, even after your changes following this ticket some examples such as Spot.TransissionAC3Ph.PowerTransfer, use synchronous frame even if inertial is selected.

BTW, @rfranke, I'm trying to write some explanation of the spot approach for my students, since it is beautiful but, for me, not everything is clear.
I will ask some minutes of you time in the next days to give me some insight on a my  very basic doubt of mine.