modelica-3rdparty/PowerSystems #25: Division by zero in average inverter

ceraolo, 2016-12-08:
Model Examples.Spot.AC3phRectifier is a good example where a diode rectifier feeds a constant source.

A twin example in the Power Systems Library feeding a  zLoadDC would be useful to have a more complete idea on the rectifier behaviour.
A first attempt I made with Rectifier was unsuccessful, so I tried with RectifierAverage.
The model I made is attached here. It causes a division by zero:
`Model error - division by zero: (1.0) / (sqrt(rectifier.iAC2)) = (1) / (0)
`
 Maybe the culprit is the RectifierAverage equation:
  `switch_dq0 = S_abs*AC.i/(sqrt(iAC2) + 1e-5);`
but this should not cause division by zero.
I will continue to dig into this issue, and in case find a solution will publish here.
However I deemed useful to to share this experience with others as well. 

The model is:
```
model RectifierAvToRL "Rectifier"

  inner PowerSystems.Utilities.System system(dynType=PowerSystems.Types.Dynamics.FixedInitial,
      refType=PowerSystems.Types.ReferenceFrame.Inertial)
    annotation (Placement(transformation(extent={{-40,40},{-20,60}})));
  PowerSystems.Blocks.Signals.TransientPhasor transPh(
    t_change=0.1,
    t_duration=0.1,
    a_start=2,
    a_end=1) annotation (Placement(transformation(extent={{-100,20},{-80,40}})));
  PowerSystems.AC3ph.Sources.Voltage vAC(use_vPhasor_in=true, V_nom=100)
    annotation (Placement(transformation(extent={{-80,0},{-60,20}})));
  PowerSystems.AC3ph.Impedances.Inductor ind(
    r=0.05,
    V_nom=100,
    S_nom=1e3) annotation (Placement(transformation(extent={{-50,0},{-30,20}})));
  PowerSystems.AC3ph.Sensors.PVImeter meterAC(
    abc=true,
    av=true,
    V_nom=100,
    S_nom=1e3,
    phasor=true,
    puUnits=false,
    tcst=0.01) annotation (Placement(transformation(extent={{-20,0},{0,20}})));

  PowerSystems.AC1ph_DC.Sensors.PVImeter meterDC(
    av=true,
    V_nom=100,
    S_nom=1e3,
    puUnits=false,
    tcst=0.01) annotation (Placement(transformation(extent={{38,0},{58,20}})));
  PowerSystems.AC3ph.Nodes.GroundOne grd1
    annotation (Placement(transformation(extent={{-80,0},{-100,20}})));

  PowerSystems.AC1ph_DC.Loads.ZloadDC zLoadDC(puUnits=false, p0=1000)
    annotation (Placement(transformation(extent={{72,0},{92,20}})));
  PowerSystems.AC3ph.Inverters.RectifierAverage rectifier
    annotation (Placement(transformation(extent={{28,0},{8,20}})));
  PowerSystems.Common.Thermal.BdCondV bdCond
    annotation (Placement(transformation(extent={{8,26},{28,46}})));
equation 
  connect(vAC.term, ind.term_p)
    annotation (Line(points={{-60,10},{-50,10}}, color={0,110,110}));
  connect(ind.term_n, meterAC.term_p)
    annotation (Line(points={{-30,10},{-20,10}}, color={0,110,110}));
  connect(transPh.y, vAC.vPhasor_in)
    annotation (Line(points={{-80,30},{-64,30},{-64,20}}, color={0,0,127}));
  connect(grd1.term, vAC.neutral)
    annotation (Line(points={{-80,10},{-80,10}}, color={0,0,255}));
  connect(meterDC.term_n, zLoadDC.term)
    annotation (Line(points={{58,10},{72,10}}, color={0,0,255}));
  connect(meterAC.term_n, rectifier.AC)
    annotation (Line(points={{0,10},{8,10}}, color={0,120,120}));
  connect(meterDC.term_p, rectifier.DC)
    annotation (Line(points={{38,10},{28,10}}, color={0,0,255}));
  connect(rectifier.heat, bdCond.heat)
    annotation (Line(points={{18,20},{18,20},{18,26}}, color={176,0,0}));
  annotation (
    Documentation(info="<html>
<p>This is a system showing the behaviour of diode rectifier component.</p>
<p>The source AC voltage is reduced during the simulation starting at around 80 ms. The DC current and AC and DC powers behave correspondingly. </p>
<p><a href=\"modelica://PowerSystems.Examples.Spot.InvertersAC3ph\">up users guide</a> </p>
</html>"),
    experiment(
      StopTime=0.2,
      Interval=0.2e-3,
      Tolerance=1e-005),
    Diagram(coordinateSystem(extent={{-100,-20},{100,60}})),
    Icon(coordinateSystem(extent={{-100,-20},{100,60}})),
    uses(PowerSystems(version="0.5.0")));
end RectifierAvToRL;

```

rfranke, 2017-01-14:
The `system` model defines fixed start values. The start values of `zLoadDC` and the Inductor `ind` are zero. One way to make the model run in Dymola is to specify steady-state initial conditions in the `system` model. Then the solver moves away from the zero.

ceraolo, 2017-01-15:
Thanks.
Is there somewhere in the embedded documentation discussion about the initial conditions? I could not find it. 
If there is not, is there a place where I could find the underlying code and try to understand things with the purpose of writing something myself?
I know that the dynType name is somewhat self-explaining, but IMO some details would be very useful.

rfranke, 2017-01-15:
`PowerSystems.Examples.Introductory` provides the best info about overall concepts, including also initialization.  Commit 72a27590ac935e40429655d3aa4992fda0c5c78f moves it up from the former `PowerSystems.Examples.Spot.Introduction`.  It really helps that you share your experience!

`Types.Dynamics` provides the following documentation:
```
Information
  FreeInitial:     Transient simulation with free initial conditions
  FixedInitial:    Transient simulation with fixed initial conditions
  SteadyInitial:   Transient simulation with steady-state initial conditions
  SteadyState:     Quasi-stationary simulation using steady-state model
```

closed 2017-01-16 by rfranke

ceraolo, 2017-01-17:
Maybe I'm too dummy, but this documentation for me does not clarify enough.

I tried the example SimulationFixedInitial, to start with.
Does "Fixed" mean that the user can choose the initial values? This seems reasonable, but the three RXLine components of the model seem no to have any way to set initial values, at least through the parameter dialog box. Or FixedInitial just means that the state variables are all zero? 

Then I tried SimulationSteadyInitial, and in this case interpretation seems clear: I should way that the initial conditions are chosen setting al the derivatives of state variables  (in  SimulationSteadyInitial the derivatives of currents) to zero.  Do you think this is correct? Are there any studies discussing when this kind of assumption involves a initial equation system that can be solved and when not?

Finally, I tried SimulationSteadyState.  I see, as stated in the "information" page that fast transients are absent. But how this is obtained? A power user wants to know how the program gets its results. I know that something like this can be attained considering complex numbers and simulating as an algebraic (on the electrical circuitry part) system, leaving the dynamics to the slower parts, such as moments of inertia. But how this relates to the dq0 transform? Other tools do this but use the Fortescue's transform, that is similar to dq0, but partly different.

Any comment or explanation or reference is welcome!