modelica-3rdparty/PowerSystems #27: Variable n_r

Rubadubby, 2016-12-31:
Hi!

some models  in OM (for example PowerSystems.AC3ph.Machines) gives an error because the n_r vairable is no declared:

[1] 11:24:56 Translation Error
[PowerSystems.AC3ph.Machines: 2208:3-2208:47]: Could not evaluate structural parameter (or constant): n_r which gives dimensions of array: L_r[n_r]. Array dimensions must be known at compile time.

[2] 11:24:56 Translation Error
Error occurred while flattening model PowerSystems.Examples.Spot.AC3ph.Machines

how can be solved? 

Thanks for this library!

rfranke, 2017-01-13:
This is a long-standing known deficiency of OpenModelica for libraries with records and inheritance. See 
[OpenModelica ticket:3302](https://trac.openmodelica.org/OpenModelica/ticket/3302). This is why the coverage of machines is quite bad. See also the [nightly tests](https://test.openmodelica.org/libraries/PowerSystems/BuildModelRecursive.html).

The OpenModelica developers are currently working with high prio on a new frontend for the translator. This will hopefully solve the issue within the next months.

closed 2018-02-08 by rfranke

tuf61179, 2019-02-22:
does anyone have any updates on the PowerSystems library? I recently just tried to instantiate a WindFarm.mo located in PowerSystems.Examples.Wind.WindFarm and also got the error of:

[1] 09:16:29 Translation Error
[PowerSystems.AC3ph.Machines: 2208:3-2208:47]: Could not evaluate structural parameter (or constant): wt1.generator.c.n_r which gives dimensions of array: L_r. Array dimensions must be known at compile time.

I'm guessing it hasn't been fixed?