modelica-3rdparty/PowerSystems #30: Erroneous initial equation when using PIline in power Transfer example

ceraolo, 2017-01-26:
I want to produce a more realistic "PowerTransfer" example in which I use realistic line parameters (the built-in OHline_132kV are unrealistic since give rise to a totally unrealistic characteristic impedance of 7.7 ohm). I want also to compare the RX line behaviour with a PI line.
Once done, I will publish here example ans thorough documentation for possible inclusion in the library.

However I came across into a nasty issue.
Consider to make a slight modification to Examples.AC3ph.Transmission.PowerTransfer, consisting in substituting the line model from RXline to PIline, and line parameters into OHline_132kV.

The model checks fine.
When run, however, the model produces the following error message:

```
The initialization problem is overspecified for variables of element type Real.
The initial equation 
0.0 = 0.0;
refers to variables, which all are known.
To correct it you can remove this equation
The initial equation 
0.0 = 0.0;
refers to variables, which all are known.
To correct it you can remove this equation
The initial equation 
0.0 = 0.0;
refers to variables, which all are known.
To correct it you can remove this equation
The initial equation 
0.0 = 0.0;
refers to variables, which all are known.
To correct it you can remove this equation

```
It seems to be more a bug than an initial condition related issue.
I will try to solve it , but I prefer to share it with GitHub, just in case someone is interested.

rfranke, 2017-02-10:
The problem is that PIline has capacitances at the connectors, i.e. voltage states, that are connected to InfBus, i.e. fixed voltages, for this example. This could possibly be treated automatically as consistently overspecified initialization problem. In the meantime, you might use a Tline for this example, i.e. inductances with current states at the connectors.

ceraolo, 2017-02-11:
ah, perfectly spotted.
I got confused by the error message.
Thanks

closed 2018-02-08 by rfranke