modelica-3rdparty/PowerSystems #35: how do I open this package under modellica?

Sinan81, 2017-08-07:
Hi, I am an absolutely new to this software, and wanted to give it a try. I am wondering if someone can tell me how to open, or get started with this package. I have openmodellica installed in my OS.

raulrpearson, 2017-08-08:
@Sinan81, I think it should be available in OMEdit through `File >> Systems Libraries`. I've checked in my OpenModelica installation (v1.12.0) and it includes what seems to be the latest version of this library.

If that weren't the case, you can always download the latest version of the library from GitHub by [cloning the repository](https://help.github.com/articles/cloning-a-repository/) or downloading a [zip of the latest commit](https://github.com/modelica/PowerSystems/archive/master.zip).

Sinan81, 2017-08-14:
Thanks. I will give it a try within few days and get back to you.

closed 2018-02-08 by rfranke