modelica-3rdparty/PowerSystems #37: On which Modelica tool has this library whiten

piliadis, 2018-06-20:
Hi,
I am interested on your work and trying to fully understand the library. But I cant run some examples (most importantly the Wind one). Both OpenModelicaEditor and SystemModeler gets out errors. I would like to know the tool you used to develop this library.
Thank you in advance,
Petros

rfranke, 2018-07-02:
The PowerSystems library has been developed with Dymola. OpenModelica supports parts of it; the WindTurbine models basing on 3ph machines don't work; see:
    https://test.openmodelica.org/libraries/PowerSystems_cpp/BuildModelRecursive.html
(These coverage tests that are linked from openmodelica.org date back to October 2017 but still reflect the current situation. New things are under development and will hopefully improve the situation soon.)

piliadis, 2018-07-02:
Thank you for your help.

closed 2018-07-02 by piliadis