modelica-3rdparty/PowerSystems #40: Please release 0.6.1 with latest fixes

casella, 2018-11-04:
@rfranke I am monitoring the [status of PowerSystems](https://libraries.openmodelica.org/branches/newInst/PowerSystems/PowerSystems.html) with the new OMC front end. There are many failures due to missing 'each' that were corrected in the master version with some PRs, but unless you make a 0.6.1 (or 0.7.0) release, Hudson won't take them into consideration, because it runs on the last released version.

Thanks!

rfranke, 2018-11-12:
Just released version 1.0.0.

closed 2018-11-12 by rfranke

casella, 2018-11-14:
Great!

dietmarw, 2018-11-14:
Just out of interest, is 1.0.0 no longer compatible with 0.6.0 or what was the reason for the major version bump?

beutlich, 2018-11-14:
We should have also updated the license (from ModelicaLicense 2 to 3-Clause BSD) for a proper v1.0.0 (which came out of sudden).