modelica-3rdparty/PowerSystems #45: problem running example model of DFIG

isablanca, 2021-07-25:
Hi all, I am totally new to the openmodelica and this powerSystem lib. When I tries to run the DFIG example, I have error as:

%%---------------------------------------------

[4] 14:55:35 Translation Error
Internal error IndexReduction.pantelidesIndexReduction failed! System is structurally singular and cannot be handled because the number of unassigned equations is larger than the number of states. Use -d=bltdump to get more information.

[5] 14:55:35 Translation Error
No system for the symbolic initialization was generated

[6] 14:55:43 Scripting Notification
PowerSystems requested package Modelica of version 3.2.2. Modelica 3.2.3 is used instead which states that it is fully compatible without conversion script needed.

[7] 14:55:43 Translation Error
[PowerSystems.AC3ph.Machines: 2208:3-2208:47]: Could not evaluate structural parameter (or constant): n_r which gives dimensions of array: L_r[n_r]. Array dimensions must be known at compile time.

[8] 14:55:43 Translation Error
Error occurred while flattening model PowerSystems.Examples.Wind.WindTurbine_DFIG

%%---------------------------------------------

do you have any suggestion to fix the problem?

rfranke, 2021-07-25:
You report an old problem with OpenModelica. It has been fixed since version 1.17 this spring. Unfortunately the example does still not work due to another problem at a later stage of model compilation:
```
Error: Internal error IndexReduction.pantelidesIndexReduction failed! System is structurally singular and cannot be handled because the number of unassigned equations is larger than the number of states. Use -d=bltdump to get more information.
Error: No system for the symbolic initialization was generated
```
This has been discussed with OpenModelica developers @casella and @kabdelhak recently. I hope that it will be solved soon.

casella, 2021-07-27:
@isablanca, I would suggest you to use 1.18.0-beta.1 if you want to run PowerSystems models