modelica-3rdparty/PowerSystems #49: Illegal call to zeros with negative argument

maltelenz, 2022-10-04:
The call to `zeros(n_q - 1)` here:
https://github.com/modelica-3rdparty/PowerSystems/blob/69fd1e7381fb50e6a0bc77d9ba168e9d6a7d9a6a/PowerSystems/Examples/AC3ph/Precalculation.mo#L1142
seems to come down to `zeros(-1)`, if we resolve `n_q`:
https://github.com/modelica-3rdparty/PowerSystems/blob/69fd1e7381fb50e6a0bc77d9ba168e9d6a7d9a6a/PowerSystems/Examples/AC3ph/Precalculation.mo#L1137
https://github.com/modelica-3rdparty/PowerSystems/blob/69fd1e7381fb50e6a0bc77d9ba168e9d6a7d9a6a/PowerSystems/Examples/AC3ph/Precalculation.mo#L1116

giving the expression `zeros(size(fill(0, 0), 1) - 1)` ==> `zeros(-1)`, which is illegal (all arguments to `zeros` need to be `>=0`), or am I missing something?

rfranke, 2023-03-21:
Just checked with OpenModelica. It generates code with zeros(0), i.e. silently corrects this during model translation. The called function doesn't use the array for n < 2.
```
function z_fromEqCirc "Calculates impedance matrix z from equivalent circuit"
  extends Modelica.Icons.Function;
  input Integer n "transient order";
...
  input SIpu.Reactance[n-1] xm2_n(each unit="1") "coupling reactance";
...
  output SIpu.Reactance[n+1,n+1] zx(each unit="1")  "impedance matrix reactive";

algorithm 
...
    for k in 2:n loop
      zx[1:k,1:k] := zx[1:k,1:k] + fill(xm2_n[k-1], k, k);
    end for;
...
end z_fromEqCirc;
```

maltelenz, 2023-03-21:
It is explicitly illegal: https://specification.modelica.org/maint/3.6/arrays.html#modelica:zeros

Wolfram System Modeler will reject models with such illegal uses.