modelica-3rdparty/PowerSystems #53: initialization problem of wind DFIG example

bww1004052196, 2023-04-29:
Hello, I am a beginner in Modelica. When I ran the wind DFIG example, the software reported an error： 
The initialization problem is inconsistent due to the following equation: 0 != 1.22474 = 1.224744871391589 * cos(select2.alpha0) * select2.v0 - inverter2.switch_dq0[1]
I am using the latest version of OpenModelica

rfranke, 2023-04-29:
Unfortunately OpenModelica is not yet able to run all models of the PowerSystems library. See also https://github.com/OpenModelica/OpenModelica/issues/8012

bww1004052196, 2023-05-02:
> Unfortunately OpenModelica is not yet able to run all models of the PowerSystems library. See also [OpenModelica/OpenModelica#8012](https://github.com/OpenModelica/OpenModelica/issues/8012) You may also have a look at the regular coverage testing results: https://libraries.openmodelica.org/branches/master/PowerSystems_latest/PowerSystems_latest.html or for Tools->Options->Simulation->TargetLanguage C https://libraries.openmodelica.org/branches/master/PowerSystems_latest/PowerSystems_latest.html and C++ https://libraries.openmodelica.org/branches/cpp/PowerSystems_latest/PowerSystems_latest.html

Thank you for your response. May I confirm if the error is related to PowerSystems.AC3ph.Machines.Asynchron_dfig? If not, I would like to use it as a reference for DFIG modeling.